ROBOTIS-GIT/dynamixel-workbench #324: Noetic release?

felixvd, 2021-02-10:
As reported in #308, the melodic-devel branch builds on Noetic. What about releasing the binaries? The same goes for dynamixel-workbench-msgs.

ROBOTIS-Will, 2021-02-15:
@felixvd 
I'm on it. Thanks!

OTL, 2021-05-19:
Hi, I came here to say +1.

ROBOTIS-Shibata, 2021-05-29:
@ROBOTIS-Will 
You know that there are situations where ROS2 users still have to use ROS1 at the same time.
The Ubuntu major version of ROS2 user is 20.04. In other words, it is Noetic in ROS1.

ROBOTIS-Will, 2021-05-31:
@ROBOTIS-Shibata 
Yes, once I release the package, it'll include all the recent ROS versions.

ROBOTIS-Will, 2021-07-30:
Hi all! 
I'm sorry about the delay in Noetic release, but it is happening now.
https://github.com/ros/rosdistro/pull/30347

yun-goon, 2025-01-13:
This issue appears to be resolved, so it will be closed.

closed 2025-01-13 by yun-goon