ROBOTIS-GIT/dynamixel-workbench #327: (with ROS) Dynamixel workbench can't find dynamixel...

judo0140, 2021-03-10:
ISSUE TEMPLATE ver. 1.0.0

Before you open issue, please refer to [ROBOTIS e-Manual](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/)

1. How to setup? (ex, U2D2, OpenCR,...)
OpenCR
ROS melodic
Dynamixel-workbench (according to e-Manual)

2. Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**

   - Model Name : XM430-W210-T

   - ID : 1

   - Baud Rate of Dynamixels : 57600

   - Protocol Version : 2.0
 
3. Write down the commands you used in order

  ```
  $ roscore    
  $ rosrun dynamixel_workbench_controllers find_dynamixel /dev/ttyACM0  (I have checked my USBport name)
  ```
 
4. Copy and Paste your error message on terminal

  ```
  SUMMARY
  ========
[ INFO] [1615358754.902633890]: Succeed to init(9600)
[ INFO] [1615358754.902686882]: Wait for scanning...
[ INFO] [1615358767.955163434]: Find 0 Dynamixels
[ INFO] [1615358767.955709794]: Succeed to init(57600)
[ INFO] [1615358767.955753130]: Wait for scanning...
[ INFO] [1615358767.957505468]: Find 0 Dynamixels
[ INFO] [1615358767.958057172]: Succeed to init(115200)
[ INFO] [1615358767.958107232]: Wait for scanning...
[ INFO] [1615358767.960077696]: Find 0 Dynamixels
[ INFO] [1615358767.960857802]: Succeed to init(1000000)
[ INFO] [1615358767.960909968]: Wait for scanning...
[ INFO] [1615358767.962606275]: Find 0 Dynamixels
[ INFO] [1615358767.963407380]: Succeed to init(2000000)
[ INFO] [1615358767.963514813]: Wait for scanning...
[ INFO] [1615358767.965253191]: Find 0 Dynamixels
[ INFO] [1615358767.966053719]: Succeed to init(3000000)
[ INFO] [1615358767.966090028]: Wait for scanning...
[ INFO] [1615358767.967743013]: Find 0 Dynamixels
[ INFO] [1615358767.968487807]: Succeed to init(4000000)
[ INFO] [1615358767.968519450]: Wait for scanning...
[ INFO] [1615358767.969737388]: Find 0 Dynamixels

5. Please, describe detailedly what difficulty you are in 

    - Dynamixel workbench can't find any dynamixel. However, in Arduino IDE, OpenCR example code (Find Dynamixel) find my dynamixel at 57600 baud rate.
    - Arduino IDE exports this image.
![image](https://user-images.githubusercontent.com/61779427/110588232-17e85600-81b8-11eb-8067-ed4e4efe153c.png)

ROBOTIS-Will, 2021-03-10:
Hi,
In order to use OpenCR as a USB-Serial interface, you should upload the `usb_to_dxl` sketch to OpenCR via Arduino IDE.
https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#opencr

judo0140, 2021-03-11:
OH MY GOD....I couldn't catch that part..
My problem is solved... very thank you!!

closed 2021-03-11 by judo0140