ROBOTIS-GIT/dynamixel-workbench #336: How to install turtlebot3 in MacOS

qixin1106, 2021-07-06:
MacOS 11.4 (20F71)
Ros2 Foxy

```
qixin@LocalQixiniMac turtlebot3-2.1.2 % colcon build                        
Starting >>> turtlebot3_description
Starting >>> turtlebot3_node
Starting >>> turtlebot3_cartographer
Starting >>> turtlebot3_example
Starting >>> turtlebot3_navigation2                                  
Starting >>> turtlebot3_teleop
--- stderr: turtlebot3_node                                                
CMake Error at CMakeLists.txt:19 (find_package):
  By not providing "Finddynamixel_sdk.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "dynamixel_sdk", but CMake did not find one.

  Could not find a package configuration file provided by "dynamixel_sdk"
  with any of the following names:

    dynamixel_sdkConfig.cmake
    dynamixel_sdk-config.cmake

  Add the installation prefix of "dynamixel_sdk" to CMAKE_PREFIX_PATH or set
  "dynamixel_sdk_DIR" to a directory containing one of the above files.  If
  "dynamixel_sdk" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< turtlebot3_node [0.81s, exited with code 1]
Aborted  <<< turtlebot3_cartographer [0.79s]
Aborted  <<< turtlebot3_description [0.85s]
Aborted  <<< turtlebot3_navigation2 [0.73s]
Aborted  <<< turtlebot3_teleop [1.00s]                                      
Aborted  <<< turtlebot3_example [1.05s]

Summary: 0 packages finished [1.56s]
  1 package failed: turtlebot3_node
  5 packages aborted: turtlebot3_cartographer turtlebot3_description turtlebot3_example turtlebot3_navigation2 turtlebot3_teleop
  1 package had stderr output: turtlebot3_node
  2 packages not processed
```

ROBOTIS-Will, 2021-07-06:
@qixin1106 
The error says that you're missing dynamixel_sdk package.
Please install the dynamixel_sdk or clone the package into your source folder before building.
Thanks.

qixin1106, 2021-07-06:
> The error says that you're missing dynamixel_sdk package.
Please install the dynamixel_sdk or clone the package into your source folder before building.
Thanks.

Thank you, I succeeded.

closed 2021-07-06 by qixin1106