ROBOTIS-GIT/dynamixel-workbench #337: The Maximum Dynamixel Control Frequency Is Limited by Delays

chcaya, 2021-07-06:
Hello,

I'm using the `DynamixelWorkbench` class to control 7 Dynamixel servos with an OpenCR and I can't control their positions at a frequency higher than 14 Hz, because of the `delay` function calls in `writeRegister` and `writeOnlyRegister` methods of the `DynamixelDriver` class.

I would like to know the reasons for these `delay` function calls, and if it would be possible to remove them?

Thank you

ROBOTIS-Will, 2021-07-07:
Hi @chcaya 

The delay in those functions seem necessary, but it is hard to track the reason behind them as it is written 3 years ago.
The Workbench is initially developed to configure DYNAMIXEL under Linux environment, therefore, it may not fully optimized for implementing an application.
If you are using OpenCR to control DYNAMIXEL, I'd recommend using using DYNAMIXEL2ARduino library or DYNAMIXEL SDK.
https://emanual.robotis.com/docs/en/software/arduino_ide/#software-installation

yun-goon, 2025-01-13:
Hi @chcaya 
Reason for delay function calls: To ensure proper communication timing and avoid conflicts.
Can delays be removed?: It is possible but may cause communication issues.

We recommend using syncWrite/syncRead as an alternative. Controlling multiple servos simultaneously reduces communication overhead and improves control frequency.
Feel free to ask for further assistance!

closed 2025-02-03 by yun-goon