ROBOTIS-GIT/dynamixel-workbench #340: ROS Noetic - Current-based Position Control Mode - position and torque control

kaltenegger, 2021-08-16:
1. Setup:

U2D2

2. Dynamixel Servo:

   - MX-106T (2.0) (1 piece)
   - ID 1
   - Baud Rate of Dynamixels 57600bps
   - Protocol Version 2.0
 
3. Starting the servo with:

```
roslaunch dynamixel_workbench_controllers dynamixel_controllers.launch
```
 
4. Error Message:

No error message
  
5. Please, describe detailedly what difficulty you are in 

I try to use the Dynamixel MX-106T (2.0) in Operating Mode 5 (Current-based Position Control Mode) with ROS Noetic on Ubuntu 20.04. Thus I am using the dynamixel-workbench on branch `noetic-devel`, The dynamixel_skd and the dynamixel_workbench_msgs are both isntalled on the master branch. To move the servo [trajectory_msgs/JointTrajectory](http://docs.ros.org/en/api/trajectory_msgs/html/msg/JointTrajectory.html) are published. Setting the effort in the [trajectory_msgs/JointTrajectroyPoint](http://docs.ros.org/en/api/trajectory_msgs/html/msg/JointTrajectoryPoint.html) does not change the servos effort. The servo is moving with maximum effort.

Is there a way to control torque (controlled via current) and position with ROS Noetic?

ROBOTIS-Will, 2021-08-17:
Hi @kaltenegger 

Thank you for your inquiry, but I'm afraid that the `dynamixel_workbench_controllers` package doesn't have an implementation for the effort at the moment.
You may try to implement your own effort controller, but DYNAMIXEL doesn't have a conversion constant value between torque and current (which DYNAMIXEL reads).
Thank you.

kaltenegger, 2021-08-17:
Thank you for the quick response.

Just for clarification: there is no way to control position and current at the same time with ROS Noetic at the moment. This would mean that Operating Mode 5 (Current-based Position Control Mode) is not compatible with ROS Noetic at this point.

ROBOTIS-Will, 2021-08-20:
@kaltenegger 
Thank you for sharing your insight in regard with this and sorry about your inconvenience.
I believe the [dynamixel_control](https://github.com/youtalk/dynamixel_control) package will cover this effort control, but it is currently being written for ROS2.

kaltenegger, 2021-08-20:
Thank you for the information.

So Operating Mode 5 (Current-based Position Control Mode) is not supported for ROS Noetic, thus the ticket can be closed.

closed 2021-08-20 by kaltenegger

suho0515, 2022-03-28:
well... I'm using dynamixel with mode 5 with dynamixel service request in ros1 noetic. so, I belive you could use operating mode 5.