ROBOTIS-GIT/dynamixel-workbench #343: Multi-motor control issues

MarqRazz, 2021-09-13:
ISSUE TEMPLATE ver. 1.0.0

1. How to setup? (ex, U2D2, OpenCR,...)
We have a setup with 24 motors connected to a Trossen Robotics 6 port hub which is then connected to a U2D2 that is plugged into our host that is running ROS2 Foxy. 
[dynamixel-workbench](https://github.com/ROBOTIS-GIT/dynamixel-workbench/tree/ros2) branch `ros2`
[dynamixel-control](https://github.com/youtalk/dynamixel_control) branch `main`
[dynamixel-workbench-msgs](https://github.com/ROBOTIS-GIT/dynamixel-workbench-msgs/tree/ros2) branch `ros2`

Image of our setup
[image]

That is we have 4 separate chains that are each connected to the 6 port hub. Each chain has the motors connected in serial (hub->motor1->motor2->motor3->motor4->motor5->motor6)

Here is a copy of our ros2_control xacro:
```
  <ros2_control name="${name}" type="system">
      <hardware>
          <plugin>dynamixel_hardware/DynamixelHardware</plugin>
          <param name="usb_port">/dev/ttyUSB0</param>
          <param name="baud_rate">2000000</param>
          <param name="joint_ids">0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23</param>
      </hardware>
  </ros2_control>
```
  
2. Please, describe detailedly what difficulty you are in 

The system is working amazing right out of the box but we have seen issues where one chain of motors can become disconnected (the wire between the 6 port hub and the first motor in the chain falls out) while executing motion which causes the motors on the other 3 chains to move out of control. 

Our question is why would removing one chain cause the others to move out of control? It looks as though the [protocol](https://emanual.robotis.com/docs/en/dxl/protocol2/) supports CRC checking but it doesn't seem to be helping or enabled. Is there an easy way to check if the CRC is being utilized? Is there anything else we can do to help make our communication more robust?

ROBOTIS-Will, 2021-09-27:
Hi @MarqRazz 
I'm sorry about the delayed response.
DYNAMIXEL should never intercept packets for other IDs unless its [Secondary ID](https://emanual.robotis.com/docs/en/dxl/x/xm540-w150/#secondaryshadow-id12) happens to be configured with the same ID number of other DYNAMIXEL.
Let me take a look at the workbench and see if there's any bug that may send unintended packets to irrelevant DYNAMIXEL.

MarqRazz, 2021-10-02:
@ROBOTIS-Will no worries I know how busy things can get!

Thanks for pointing me to the Secondary ID I will have our customer check what that have it set to. I went over their wire routing and made it much more robust which has helped suppress the issue. If we continue to have problems I will post more updates here.

Thanks for your help!

yun-goon, 2025-01-13:
This issue will be closed due to prolonged inactivity. If the issue persists, feel free to reopen it.

closed 2025-01-13 by yun-goon