ROBOTIS-GIT/dynamixel-workbench #345: How can I set Extended Position Control Mode to XM or XH Series?

yamamotoshota0903, 2021-10-01:
ISSUE TEMPLATE ver. 1.0.0

Before you open issue, please refer to [ROBOTIS e-Manual](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/)

1. How to setup? (ex, U2D2, OpenCR,...)
setup with U2D2.

2. Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**

   - Model Name:XM540-W270-R

   - ID:11

   - Baud Rate of Dynamixels:57600

   - Protocol Version:2.0?

3. Please, describe detailedly what difficulty you are in 

    - I want to control dynamixel(XMorXHseries) in Multi-Turn(Extended Position Control)Mode.
    - I searched dynamixel_workbench_toolbox/exampled/src for example code, but I can not find example about Extended Position Control Mode.
    - I searched dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_workbench.cpp for method, and I found that I can't use ExtendedPositionControl without dynamixelpro in ControlModeDynamixelWorkbench::setOperatingMode.
    - Instead of Extended Position Control Mode, I tried Current Based Position Control Mode. However, It can't move without sleep(). (I want to use it without sleep().) And it's initial speed is too slow. 
   
####my question
- Can I use Extended Position Control Mode with XM and XH Series? or Can I use Dynamixel_workbench_toolbox to set Extended Position Control Mode?
- Is it any way to use Current Based Position Control Mode more useful?
- I'm sorry that my discription is not clear.

ROBOTIS-Will, 2021-10-06:
Hi @yamamotoshota0903 

You may use the setExtendedPositionControlMode function to operate the X series DYNAMIXEL in multi turn mode.
For example, you can replace the [jointMode() function](https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/master/dynamixel_workbench_toolbox/examples/src/h_Position.cpp#L69) of the h_Position.cpp as below.

```cpp
result = dxl_wb.setExtendedPositionControlMode(dxl_id, &log);
```

Unlike the jointMode(), the setExtendedPositionControlMode doesn't automatically turn on the torque, so you should add additional torqueOn() after changing the control mode.
```cpp
dxl_wb.torqueOn(dxl_id, &log);
```

The Current based Position Control mode is useful when current control is important while operating under position mode such as gripper.

Thank you.

yamamotoshota0903, 2021-10-12:
Thanks a lot!!
I closed issue.

closed 2021-10-12 by yamamotoshota0903