ROBOTIS-GIT/dynamixel-workbench #347: Support for ros2-galactic: colcon build error

3mrrrx, 2021-11-18:
basic complication unsing colcon on ros2-galactic results in errors on both branches:

 - ros2-devel
 - ros2 

`
source /opt/ros/galactic/setup.bash
rosdep install --from-paths src --ignore-src -r -y
colcon build --symlink-install --cmake-args -DCMAKE_EXPORT_COMPILE_COMMANDS=ON
`

error:
`
fatal error: hardware_interface/base_interface.hpp: No such file or directory
   20 | #include <hardware_interface/base_interface.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
`

3mrrrx, 2021-11-18:
these  the available libraries in for both galatic and foxy.

 `
ros-galactic-dynamic-edt-3d                          ros-galactic-dynamixel-sdk-custom-interfaces-dbgsym
ros-galactic-dynamic-edt-3d-dbgsym                   ros-galactic-dynamixel-sdk-dbgsym
ros-galactic-dynamixel-sdk                           ros-galactic-dynamixel-sdk-examples
ros-galactic-dynamixel-sdk-custom-interfaces         ros-galactic-dynamixel-sdk-examples-dbgsym

ros-foxy-dynamic-edt-3d                          ros-foxy-dynamixel-sdk
ros-foxy-dynamic-edt-3d-dbgsym                   ros-foxy-dynamixel-sdk-custom-interfaces
ros-foxy-dynamic-graph                           ros-foxy-dynamixel-sdk-custom-interfaces-dbgsym
ros-foxy-dynamic-graph-dbgsym                    ros-foxy-dynamixel-sdk-dbgsym
ros-foxy-dynamixel-hardware-interface            ros-foxy-dynamixel-sdk-examples
ros-foxy-dynamixel-hardware-interface-dbgsym     ros-foxy-dynamixel-sdk-examples-dbgsym
`

apperently ros-galactic-dynamixel-hardware-interface doesn't exist!!!

any ideas how to fix this?

ROBOTIS-Will, 2021-12-07:
Hi @3mrrrx 

Since this issue is being discussed in the original repository, I'll close this thread.
https://github.com/youtalk/dynamixel_control/issues/15
Thanks!

closed 2021-12-07 by ROBOTIS-Will