ROBOTIS-GIT/dynamixel-workbench #348: XC-330-T288-T support

palwanishan, 2021-11-27:
I want utilize XC-330-T288-T as manipulator gripper servo and control with ROS. But dynamixel_workbench doesn't support this model somehow. 

rosrun dynamixel_workbench_controllers find_dynamixel /dev/ttyUSB0

This command gives me result in below:

id : 28, model name : (null)
id : 29, model name : (null)

Instead of (null) it is supposed to print the model name such as XC-330-T288-T.

To be specific I modified my Open-manipulator-X and made it 7-dof including E.E. with XC-330-T288-T.

ROBOTIS-Will, 2021-12-07:
Hi @palwanishan 

I'm sorry about the delayed response.
The X330 series has been updated in the `develop` branch #349.
Please try with it and let us know if it works for you.
Once I get a confirmation from you, I'll merge it into the master.
Thanks!

palwanishan, 2021-12-07:
Greetings! I appreciate your response. 
I did exactly you told and here is progress. Now dynamixel_workbench recognizing XC-330. 

But still have some problem:

[ERROR] [DynamixelWorkbench] Failed to set velocity based profile!
[ERROR] [DynamixelTool] Can't find Item
[ERROR] [DynamixelWorkbench] Failed to change torque status!

I attached screen captures as well.

I'm using ROS-Noetic

Thanks!

[image]
[image]
[image]

palwanishan, 2021-12-07:
I looked up source code and copy pasted 1 line and it's working now.

dynamixel_item.cpp - 
const ControlItem *DynamixelItem::getControlTable(uint16_t model_number) {
else if (num == XL330_M077 || num == XL330_M288 || num == XC330_M181 || num == XC330_M288 || num == XC330_T181 || num == XC330_T288)}

previously it was - 

const ControlItem *DynamixelItem::getControlTable(uint16_t model_number) {
else if (num == XL330_M077 || num == XL330_M288)}

Thanks!

ROBOTIS-Will, 2021-12-08:
Oh, thanks for testing and reviewing the code.
I've appended the missing models in the getControlTable function and committed to the `develop` branch(#349).
I'll merge and release this update for melodic and noetic within a couple weeks.
Thank you!

PR #349 (2021-12-08): Develop
- fix #348
- revert the vector type migration in #338 as it doesn't work with OpenMANIPULATOR-X and cannot be fully tested at the moment.
closed 2021-12-08 by ROBOTIS-Will