ROBOTIS-GIT/dynamixel-workbench #354: PWM calculation using PID control

eshafeeqe, 2022-04-04:
ISSUE TEMPLATE ver. 1.0.0

Before you open an issue, please refer to [ROBOTIS e-Manual](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/)

1. How to set up? (ex, U2D2, OpenCR,...)
OpenCR

3. Which Dynamixel have you used? and how many? **(Please describe the below format to all connected Dynamixels)**

   - Model Name: XH430-W210

   - ID : 0

   - Baud Rate of Dynamixels : 4 Mbps

   - Protocol Version 2.0
 
4. Write down the commands you used in order

  ```
  $ roslaunch single_leg_pkg ros52_pd_control.launch
  ```
 
5. Copy and Paste your error message on the terminal

  ```
  SUMMARY
  ========

  PARAMETERS
   * /baud_rate: 4Mbps
   * /device_name: /dev/ttyACM0
   * /ping: True
   * /ping_id: 0
   * /rosdistro: noetic
   * /rosversion: 1.12.13
   * /scan_range: 10

  NODES
    /
      single_dynamixel_controller (dynamixel_workbench_single_manager/single_dynamixel_controller)
      single_dynamixel_monitor (dynamixel_workbench_single_manager/single_dynamixel_monitor)

  auto-starting new master
  process[master]: started with pid [8594]
  ROS_MASTER_URI=http://192.168.1.3:11311

  setting /run_id to de3bd71e-728e-11e8-9f66-d8cb8af3d300
  process[rosout-1]: started with pid [8607]
  started core service [/rosout]
  process[single_dynamixel_monitor-2]: started with pid [8618]
  process[single_dynamixel_controller-3]: started with pid [8626]
  [PortHandlerLinux::SetupPort] Error opening serial port!
  ================================================================================REQUIRED process [single_dynamixel_monitor-2] has died!
  process has died [pid 8618, exit code -11, cmd /home/darby/catkin_ws/devel/lib/dynamixel_workbench_single_manager/single_dynamixel_monitor __name:=single_dynamixel_monitor __log:=/home/darby/.ros/log/de3bd71e-728e-11e8-9f66-d8cb8af3d300/single_dynamixel_monitor-2.log].
  log file: /home/darby/.ros/log/de3bd71e-728e-11e8-9f66-d8cb8af3d300/single_dynamixel_monitor-2*.log
  Initiating shutdown!
  ================================================================================
  [single_dynamixel_controller-3] killing on exit
  [single_dynamixel_monitor-2] killing on exit
  [rosout-1] killing on exit
  [master] killing on exit
  shutting down processing monitor...
  ... shutting down processing monitor complete
  ```
  
6. Please, describe detailedly what difficulty you are in 

I am trying to understand how the Dynamixal estimates the PWM voltage while using PID control. I set 

**PID Controller**

KpP(84) = 640, 
KpD(84) = 0
KpI(84) = 0

**Feed Forward Controller**

KFF1st  = 0
KFF2nd = 0

Also set  Profile Velocity (108) and Profile Acceleration as zero. 

In my current understanding, the PWM value is calculated as 

PWM = kp*(position_error)

In order to verify this, the target position is set to 60 degrees (started from zero) and we collected the Current PWM and Current Position for one second.  As we have some confusion about up here whether it is 640 or 640/128 (as explained in the XH430-W210 datasheet shown below.) 

[image]

However with the data when we calculate PWM using the equation  (PWM = kp*(position_error)) we got different data from the PWM data we got from the Dynamixel readings. 

The comparison curve with kp = 640 is shown below. 

[image]

The comparison curve with kp = 640/128 = 5 is shown below. 

[image]

However, we got more close matches when we use kp = 40 as shown below. 

[image]

We are doing this experiment for validating whether the PID controller actually working as expected since these gains are very important for the applicability of our controller in the real system. However, we are getting results that do not align with our calculation. 

Can you explain why there is a discrepancy in the PWM and the kp value here and anything else affecting the PWM estimation? 
Any help in regards greatly appreciated. 

Thanks and Regards, 
Shafeeq E

ROBOTIS-Will, 2022-04-21:
Hi @eshafeeqe 

I'm sorry about the delayed response on this issue.
The Goal PWM in the Position Controller of DYNAMIXEL works as a PWM limiter which means the it only cuts off the output from the PID controller at its set value.
[image]

As you can see in the diagram above, the PWM is not directly involved in the PID controller as the PID is calculated based on the Present Position of the previous frame.

The larger P gain term in PID controller will generate more power when the actual(present) position is closer to the goal position like the  image of Kp = 640.
On the other hand smaller P gain exerts smaller power to compensate the gap between the actual(present) and goal positions, therefore, the voltage is lower than the first one.

I noticed is that regardless of the Kp value, the Current Position for three cases seem identical.
How these graphs / data are recorded and measured?

eshafeeqe, 2022-04-26:
Hi @ROBOTIS-Will 

Thanks for the response.  We are trying to understand how the dynamixel estimates PWM while using the P controller in the experiment above. In my understanding the PWM = Kp*position_error. However, we are getting different information from the dynamixel readings. 

We used the OpenCR board and used dynamixel workbench API to get and set information in Dynamixel. 
We executed the following steps in the above experiment. 

1) Initialize dynamixel in extended position mode.
2) Set Kp = 640
3) Reset the motor to zero position using setG

We reset the dynamixel to zero.

closed 2022-04-26 by eshafeeqe