ROBOTIS-GIT/dynamixel-workbench #358: Stop motor movement after a communication error

alexandremstf, 2022-07-15:
Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**

   - Model Name: MX64
   - ID: 1
   - Baud Rate of Dynamixels: 57600
   - Protocol Version: 1.0

Write down the commands you used in order
  ```
  $ roscore    
  $ roslaunch dynamixel_workbench_controllers dynamixel_controllers.launch
  ```
 
Is there some way to stop motor after some error? In my case i was using the motor in wheel mode, and the usb disconnected from the U2D2 and the motor keeps moving. I can't find some way to configure this stop. Someone had this problem?

ROBOTIS-Will, 2022-07-21:
Hi @alexandremstf 

The MX-64 with DYNAMIXEL Protocol 2.0 support this feature with [BUS watchdog](https://emanual.robotis.com/docs/en/dxl/mx/mx-64-2/#bus-watchdog).
By default, the MX-64 comes with Protocol 1.0 firmware, but you can upgrade it with the Protocol 2.0 from DYNAMIXEL Wizard 2.0 by uploading the MX-64(2.0) firmware.
Please refer to the DYNAMIXEL Wizard 2.0 firmware recovery section of the eManual below.
https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/
Thanks!

alexandremstf, 2022-08-19:
Many thanks Will, worked like a charm, now when something like a communication error happens the motors stop and my robot still in safe.

closed 2022-08-19 by alexandremstf