ROBOTIS-GIT/dynamixel-workbench #361: Release for ROS2?

ijnek, 2022-09-20:
Thank you for providing this package, I'm wondering if there are any plans on releasing this into ROS 2?

youtalk, 2022-09-24:
We would like to the release package.  Thank you in advance.

ROBOTIS-Will, 2022-09-26:
Hi all,
This package has not been released for ROS2 yet.
Could you specify the ROS2 distro you are working on?

ijnek, 2022-09-26:
If releasing the package into all active distros is an option, that'll be the best. I've been contributing to https://github.com/youtalk/dynamixel_control, which supports foxy, galactic, humble and rolling, but [can't be released](https://github.com/youtalk/dynamixel_control/issues/29), as it depends on dynamixel-workbench.

ROBOTIS-Will, 2022-10-13:
Due to the changes in releasing new ROS2 package, I'm waiting for OR to create new release repositories for `dynamixel_workbench` and its `msgs` repo.
I'll move to the next step as soon as the release repositories are created in the ros2-gbp.

https://github.com/ros2-gbp/ros2-gbp-github-org/pull/136

Thank you.

ROBOTIS-Will, 2022-10-27:
@ijnek @youtalk 
The PR was merged!
I'll start working on releasing both repos.
Thanks for your patience and it was my absolute pleasure to meet you guys in person during ROSCon :)

ijnek, 2022-10-27:
Thanks Will, it was my pleasure to meet you both too! I've been tracking the PRs over on rosdistro and ros2-gbp and I understand that there has been difficulty due to the changes in the release procedure, but I really appreciate you being active on this!

ps. I think ROBOTIS had the coolest booth at ROSCon

youtalk, 2022-10-30:
Great work! Thank you for your effort. We can move forward.
We love Dynamixel!

ijnek, 2022-12-09:
Another big thanks @ROBOTIS-Will! Closing this off as completed.

closed 2022-12-09 by ijnek