ROBOTIS-GIT/dynamixel-workbench #369: [TxRxResult] Incorrect status packet!

YulsXDP, 2023-03-16:
Hi i tryed to use the motors of dynamixel Ax 12A whit the controler CM 530, i watched many tutorials and instructions using the following links from Robotis
https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/python_linux/
https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/
https://emanual.robotis.com/docs/en/software/rplustask3/
https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/matlab_windows/

the only thing that works was R+ Task 3.0 for moving the motor, i'm new using the motors, i tryed change the plataform (matlab to python) but the error is constant "[TxRxResult] Incorrect status packet!"

I'm use
   - Dynamixel AX-12A

   - ID 15

   - 1000000 or 57600

   - Protocol Version 1.0
 
3. the commands i used
for python dowload SDK controler of Dynamixel and AX12A Controler of the user @aakieu whit this tutorial
https://www.youtube.com/watch?v=SqKS_cFuPh0&t=50s
changed the com por and baud rate 

  ```
from Ax12 import Ax12

# e.g 'COM3' windows or '/dev/ttyUSB0' for Linux
Ax12.DEVICENAME = 'COM3'

Ax12.BAUDRATE = 57_600

# sets baudrate and opens com port
Ax12.connect()

# create AX12 instance with ID 10 
motor_id = 15
my_dxl = Ax12(motor_id)  
my_dxl.set_moving_speed(200)

def user_input():
    """Check to see if user wants to continue"""
    ans = input('Continue? : y/n ')
    if ans == 'n':
        return False
    else:
        return True

def main(motor_object):
    """ sets goal position based on user input """
    bool_test = True
    while bool_test:

        print("\nPosition of dxl ID: %d is %d " %
              (motor_object.id, motor_object.get_present_position()))
        # desired angle input
        input_pos = int(input("goal pos: "))
        motor_object.set_goal_position(input_pos)
        print("Position of dxl ID: %d is now: %d " %
              (motor_object.id, motor_object.get_present_position()))
        bool_test = user_input()

# pass in AX12 object
main(my_dxl)

# disconnect
my_dxl.set_torque_enable(0)
Ax12.disconnect()

  ```
 
4. the error message on terminal

Succeeded to open the port
Succeeded to change the baudrate
[TxRxResult] There is no status packet!
[TxRxResult] Incorrect status packet!
Moving speed of dxl ID: 15 set to 0
[TxRxResult] Incorrect status packet!

Position of dxl ID: 15 is 0
goal pos:
  ```
  SUMMARY
  ========

  PARAMETERS
   * /baud_rate: 57600
   * /device_name: COM3
   * /ping_id: 15
   * /python versoin: 9.9.54

ROBOTIS-Will, 2023-03-22:
Hi @YulsXDP 

The CM530 is an embedded controller that does not support DYNAMIXEL SDK.
Since the R+Task 3.0 works with the built in CM530 firmware, it would be the easiest way to operate the AX-12A.

If you want to use the CM530 as a customized controller, please see Embedded C section of the eManual
https://emanual.robotis.com/docs/en/software/embedded_sdk/embedded_c_cm530/

If you are new to DYNAMIXEL, I would recommend starting with the `DYNAMIXEL Starter Set` so that you can try operating various DYNAMIXEL using `DYNAMIXEL Wizard 2.0` or `DYNAMIXEL SDK`.

shiba-8ro, 2023-03-24:
If you want to do full-scale development using Dynamixel SDK, I recommend OpenRB-150 instead of CM-530.
https://www.youtube.com/watch?v=QHyzqY2iRWA

DDGC16, 2024-05-07:
@ROBOTIS-Will Hi, so does that mean it wont work on the CM-5 controller as well?

shiba-8ro, 2024-05-08:
@DDGC16 
Hi, CM-5, and CM-510 are older boards than CM-530. Above all, CM-5 cannot be developed with Arduino IDE. Boards newer than CM-530 and compatible with the Dynamixel SDK are OpenCM904, OpenCR1.0, and OpenRB-150. OpenCM904 is a discon. OpenCR1.0 has high performance but is expensive. I once again recommend the use of OpenRB-150.

yun-goon, 2025-01-22:
This issue will be closed due to prolonged inactivity. If the issue persists, feel free to reopen it.

closed 2025-01-22 by yun-goon