ROBOTIS-GIT/dynamixel-workbench #370: Incorrect status packet

gyxx323, 2023-03-19:
I use U2D2 connection, when I connect to the PC everything is fine.
But when I use Jetson to connect to U2D2, sometimes everything is fine, sometimes it gives an error: Incorrect status packet. And it sends a series of trajectory points that it wants the servo to execute, part of which it will execute intermittently.

The servos I use are XM430, XC430 and XL430
Baud rate: 51200
The period of sending the desired position of the servo: 50Hz

[image]

ROBOTIS-Will, 2023-03-22:
The baudrate of 57600 bps doesn't seem fast enough for 50Hz control frequency.
Please try with higher baudrate, and also if possible, please try reading the data packet with DYNAMIXEL Wizard 2.0 packet analyzing feature.

gyxx323, 2023-03-24:
What is the recommended baud rate for 50hz control? Also, does voltage instability affect this(The motor has not been cut off)?
Thanks!

shiba-8ro, 2023-03-24:
@gyxx323 
U2D2 and Dynamixel X series support up to 4.5Mbps. you're worth to try it.
Note that USB (FTDI chip) has latency.
https://granitedevices.com/wiki/FTDI_Linux_USB_latency

gyxx323, 2023-03-26:
Thank you. I will try it!

gyxx323, 2023-03-26:
> @gyxx323 U2D2 and Dynamixel X series support up to 4.5Mbps. you're worth to try it. Note that USB (FTDI chip) has latency. https://granitedevices.com/wiki/FTDI_Linux_USB_latency

Thank you. I will try it!

yun-goon, 2025-01-13:
This issue will be closed due to prolonged inactivity. If the issue persists, feel free to reopen it.

closed 2025-01-13 by yun-goon