ROBOTIS-GIT/dynamixel-workbench #372: No executable found

Nielk7991, 2023-05-11:
1. How to setup? (ex, U2D2, OpenCR,...)
U2D2
2. Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**

   - Model Name
Dynamixel MX-28
   - ID
1
   - Baud Rate of Dynamixels
57600
   - Protocol Version
 1.0
3. Write down the commands you used in order

  ```
cd dm_ws/src
  ```
git clone -b humble-devel https://github.com/ROBOTIS-GIT/dynamixel-workbench.git
  ```
cd ..
  ```
colcon build
  ```
source ~/.bashrc
  ```
ros2 run dynamixel_workbench_toolbox o_Find_Dynamixel
  ```
 
4. Copy and Paste your error message on terminal

  ```
No executable found

  ```
  
5. Please, describe detailedly what difficulty you are in 

I cloned the Dynamixel SDK, Dynamixel workbench and the dynamixel workbench msgs for ros2 humble. I build the packages and the environment. 
When I try to run a Node like `ros2 run dynamixel_workbench_toolbox o_Find_Dynamixel` it didnt find any executables.

Thanks

kneave, 2023-06-01:
Did you get a solution to this please? I'm trying to migrate from ROS1 to ROS2 Humble and it looks like the documentation for workbench hasn't been updated for ROS2

Nielk7991, 2023-06-02:
No, unfortunaly not. I`m just waiting for an answer. 
Can you share your code if you have migrate it to ROS2 Humble?

kneave, 2023-06-02:
Sadly I haven't any, I've decided to ditch ROS2 for now and stick with Noetic a bit longer.

Elston10, 2023-07-02:
has any one found a solution for this

Elston10, 2023-07-02:
i have tried a lot of different ways of solving this but i still cant fix this

ARLunan, 2024-04-03:
After installing dynamixel-workbench packages on my Unbuntu 22.04/ROS 2 Humble, a set of dynamixel packages could be listed that are located in the ros2 /opt/ros/humble/share folder. However, I too discoverd that there are no executables or launch files available. Also the Current Robotis eManual for Dynamixel Workbench (https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#ros-tutorials) only has ros commands (which in pending EOL), not ROS 2. So this eManual should be updated to reflect the latest available ROS Distro..

shiba-8ro, 2024-04-16:
ROBOTIS and users close to them know that it is common sense to use the DYNAMIXEL SDK for ROS 2, but I also think that it is necessary to make an announcement in the e-manual that many people view.
https://youtu.be/E8XPqDjof4U?si=xyvWUL9bxcHkI35o

yun-goon, 2025-01-13:
We apologize for the delayed response.
To address DXL issues, we have assembled a dedicated TF team and are actively working on resolutions.

The examples in workbench_toolbox cannot be executed using ROS. However, they can be used after building them as standalone applications. Here is how to run Dynamixel-Workbench on Linux without ROS:

Download DynamixelSDK

```
git clone https://github.com/ROBOTIS-GIT/DynamixelSDK
```
Build DynamixelSDK

Follow the instructions provided in the e-Manual:
[Build the Library](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/cpp_linux/#build-the-library)

Download Dynamixel-Workbench

```
git clone https://github.com/ROBOTIS-GIT/dynamixel-workbench
```
Build Dynamixel-Workbench and Run Examples

```
$ cd ${YOUR_DOWNLOAD_PATH}/dynamixel_workbench/dynamixel_workbench_toolbox/examples
$ mkdir build && cd build
$ cmake ..
$ make
$ sudo chmod a+rw ${USB_PORT}
$ ./monitor
```
Let us know if you need further assistance!

closed 2025-02-03 by yun-goon