ROBOTIS-GIT/dynamixel-workbench #375: Joint_trajectory control does not respond to velocity commands

echoGee, 2023-07-20:
ISSUE TEMPLATE ver. 1.0.0

Before you open issue, please refer to [ROBOTIS e-Manual](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/)

1. How to setup? 
- U2D2

4. Which Dynamixel have you used?
   - Model Name
XM540-W150-R
 
5. Write down the commands you used in order

  ```
  $ roscore    
  $ roslaunch dynamixel_workbench_controllers dynamixel_controllers.launch use_moveit:=true
  $ rosrun customcontroller run_controller
  ```
  
6. Please, describe detailed what difficulty

We send a trajectory with joint position and velocity in the joint_trajectory topic. But the servo does not respond to velocity. We have tried setting https://emanual.robotis.com/docs/en/dxl/x/xd540-t150/#operating-mode11 to velocity mode. But still doesn't work

echoGee, 2023-10-14:
Any update on this pls ?

yun-goon, 2025-01-16:
Hi @echoGee 
Sorry for the late reply.
The issue seems related to velocity commands not being processed by the Dynamixel. Here are some steps to resolve the problem:

### 1. Check Operating Mode
Ensure the Dynamixel is set to Velocity Control Mode. Use the following commands to set and reboot:

```rosservice call /dynamixel_workbench/dynamixel_command "command: ''
id: 1
addr_name: 'Operating_Mode'
value: 1"
```
```
rosservice call /dynamixel_workbench/dynamixel_command "command: 'reboot' id: 1"
```
### 2. Verify joint_trajectory Topic
Ensure the velocity field in the topic contains valid data:

```
rostopic echo /joint_trajectory
```
### 3. Test Basic Velocity Command
Send a direct velocity command to verify hardware functionality:
```
rosservice call /dynamixel_workbench/dynamixel_command "command: ''
id: 1
addr_name: 'Goal_Velocity'
value: 100"
```
### 4. Check Firmware
Ensure the Dynamixel firmware is updated to the latest version using the Dynamixel Wizard.

### 5. Provide Details About customcontroller
If the above steps do not resolve the issue, it would be helpful to have more information about the customcontroller code you are using. Specifically:

How it processes velocity commands from the joint_trajectory topic.
Whether it correctly communicates these commands to the Dynamixel.
This additional context will help in identifying whether the problem lies in the control logic or hardware setup. 😊

closed 2025-02-03 by yun-goon