ROBOTIS-GIT/dynamixel-workbench #376: Newer Dynamixel models not properly supported in ROS2 branches

sven-hoek, 2023-08-29:
## 1. How to setup? (ex, U2D2, OpenCR,...)
Connected to the Dynamixels via RS485 over a USB->RS485 converter (Waveshare), then RS485 to TTL via Dynamixel Communication Bridge.

## 2. Which Dynamixel have you used? and how many? **(Please describe below format to all connected Dynamixels)**

   - XC330-288T
   - ID 10
   - 115200 baud
   - Protocol Version 2.0
 
## 3. Write down the commands you used in order

I'm using the workbench as a library, calling

```cpp
m_dynamixel_workbench.setCurrentControlMode(joint.dxl_id, &log)
```
Where `joint.dxl_id` is my Dynamixel's ID, of course.

## 5. Please, describe detailedly what difficulty you are in 
I'm on branch `ros2` (using it in Humble). When I want to switch to *Current Control Mode*, nothing happens (no comms on the bus). Replacing the call to `setCurrentControlMode()` with a call to `setExtendedPositionControlMode()` in the same place, it successfully changes to the *Extended Position Control Mode*. So I went into the workbench's code and saw that the checks don't include the XC330 models and probably just need to be added as in https://github.com/ROBOTIS-GIT/dynamixel-workbench/commit/99f377b7064f56093c350ffd619897c53f7e608f  :
https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/140ddf6566fd01d1fdbbcbd201ccc5bddfc7a907/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_workbench.cpp#L627-L637

It looks like commit https://github.com/ROBOTIS-GIT/dynamixel-workbench/commit/c9e4c756a2c222475372f390d5396b9155693b98 adds some support to the ROS2 branches but misses the above mentioned checks/

The `master` branch has commit https://github.com/ROBOTIS-GIT/dynamixel-workbench/commit/99f377b7064f56093c350ffd619897c53f7e608f which seems to add full support for them -> can it possibly be cherry-picked onto the ROS2 branches?

The issue https://github.com/ROBOTIS-GIT/dynamixel-workbench/issues/278 (more than 3 years old by now) could be sort of related and would possibly be resolved when resolving this issue.

sven-hoek, 2023-10-04:
Similarly, the selection of the current conversion factor in `convertCurrent2Value()` and `convertValue2Current()` doesn't have the correct conversion factor for some models, at least not for the one mentioned above. The latter seems to be fixed on the master branch, but not on the ROS2 branch:
https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/67a163199cab651fdf4aceb984f3b75ea30847fe/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_workbench.cpp#L1363-L1395
https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/140ddf6566fd01d1fdbbcbd201ccc5bddfc7a907/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_workbench.cpp#L1367-L1406

soham2560, 2024-05-11:
any update on this?