ROBOTIS-GIT/dynamixel-workbench #382: xm430-w350 overload error

is3rlab, 2024-02-22:
Hi everyone,

[image]

I currently on a WidowX-250 6DOF robot arm. When I test it, the motor 6 shows overload error. When I used the Dynamixel Wizard tool reboot the motor or unplug & replug the power, the error and red light will dismiss. However, when I try it again, the error and red light will come back again. Please check the below picture:
[image]
[image]

So can anyone help me solve this?

Thank you very much for your help!

Best.

shiba-8ro, 2024-02-23:
@is3rlab 
Things you can try right away.
1. Check that the tips of the screws that secure the frame to the servo horn and idler are not touching the servo case and causing a load.
2. The washer attached to the neck of the servo horn often fails to attach and becomes a load.
3. FW recovery https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#firmware-recovery
If that doesn't work, unfortunately, that XM430 will require hospitalization.

shiba-8ro, 2024-02-24:
@is3rlab 
Looking at the photo, you may not have installed the servo horn correctly.
Check that the mark on the output shaft of the XM430 matches the mark on the servo horn.

yun-goon, 2025-01-20:
Hi @is3rlab 
Please check if the frames and joints of the robot arm are properly connected and aligned to avoid unnecessary load on the motor. Additionally, try the following steps:

1. Firmware Update: Ensure that the firmware of the XM430-W350 is up to date using the Dynamixel Wizard tool.
2. Factory Reset: Perform a factory reset on the motor to restore default settings and resolve any potential configuration issues.
If the issue persists after these steps, it might indicate a hardware problem with the motor. In that case, we recommend sending the motor in for repair or replacement.

yun-goon, 2025-02-03:
Please try the solutions mentioned above.
If the same issue persists, try replacing the Dynamixel with another one if possible and check if the same symptoms occur.
If the problem is with the Dynamixel itself, we recommend requesting after-sales service (AS) for support.
If you need further assistance, please feel free to reopen this issue.

closed 2025-02-03 by yun-goon

yun-goon, 2025-02-13:
Hi, while doing some further research I discovered something interesting. There’s a feature called self-diagnosis that allows you to check if there are any issues with your Dynamixel. Please take a look at the details here:

https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#self-diagnosis

Could you try running the self-diagnosis to see if everything is functioning correctly?