ROBOTIS-GIT/dynamixel-workbench #384: Reason for removal of support for Dynamixel models in release 2.2.3

nilp-dromeda, 2024-07-24:
I am currently using the newest release `2.2.3` to interact with Dynamixels that we are using. One of the key requirements was to be able to enable Current based position control on the dynamixels. We use a few different versions of dynamixels and most of them were enabled to position control, however the `dynamixel-workbench-toolbox` did not allow for the two models shown below:

- `XL330-M288-T`
- `XC330-T288-T`

Upon looking at the source code for release `2.2.3` I realised that a lot of the supported Dynamixel models were removed from `dynamixel_workbench.cpp` relative to the `master` branch. I have included a diff file which details the the `if condition` checks that had a variety of models removed from them (patch file not supported for upload on linux):

[Git diff between `2.2.3` and `master`](https://github.com/user-attachments/files/16368212/additional_motors.txt)

What is the reason for these motor models not being added from main into this release?

pazeshun, 2024-12-05:
Now `master` and `develop` branches are deleted, so we have no way to move `XL330`/`XC330` actuators on ROS Noetic with this repository.
[The minimum necessary commit](https://github.com/ROBOTIS-GIT/dynamixel-workbench/commit/99f377b7064f56093c350ffd619897c53f7e608f) does not belong to any branch on this repository.
@robotpilot Do you have any plan to recover from this situation?

robotpilot, 2024-12-05:
@pazeshun 
I sincerely apologize for any inconvenience caused.
For a long period, maintenance of open-source codes like the Dynamixel SDK and Workbench has not been carried out. As we organized the previous codes under a new plan and created new branches for future development, this issue arose. This was my oversight. I have temporarily restored the master and develop branches you mentioned. In the future, these branches will be better managed under new branches. Once again, I sincerely apologize for any inconvenience.

pazeshun, 2024-12-05:
@robotpilot 
Thank you very much for your support.
I found `master` branch lost one commit:
https://github.com/ROBOTIS-GIT/dynamixel-workbench/commit/67a163199cab651fdf4aceb984f3b75ea30847fe

Also, we have some PRs which are automatically closed by the branch deletion:
https://github.com/ROBOTIS-GIT/dynamixel-workbench/pull/385
https://github.com/ROBOTIS-GIT/dynamixel-workbench/pull/386
https://github.com/ROBOTIS-GIT/dynamixel-workbench/pull/379

Will `In the future, these branches will be better managed under new branches` take long time?
If so, we will re-create those PRs against current `master`.

robotpilot, 2024-12-05:
@pazeshun Thank you very much for your support. I reopen the you mentioned PRs. Thanks!