ROBOTIS-GIT/dynamixel-workbench #395: ViperX 300S Joint detection error

Ottiehoblen, 2025-04-21:
[image]

Hi, we're having issues with our robotic arm, our software cannot launch because it cannot detect all joints except joint 1. 

The arm did have a minor impact with the table but we don't think that should ruin any motor or cable connections. For instance, we chekt and there is no visual damadge on the gripper nor any other parts, and all cables appears to be sitting where they are meant too. Although we haven't disassembled the arm and chekt. 

Earlier before we also had a minor impact(2 of total as far) and got that Arvis couldn't find any joints. Therefore we erased all libraries and rebuilt the workspace with catkin and the error message disapeard and the arm worked. So we tried it again and it didnt work. 

Any tips on what could be the error and how to solve it?

Best
Alex

yun-goon, 2025-04-21:
Hello Alex,

If the DYNAMIXELs are not being recognized after the impact, there's a high possibility of an overload error. If the DYNAMIXEL LEDs are blinking and the motors are not being detected, you can connect to DYNAMIXEL Wizard 2.0 (https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/) to check the specific error. If there is an error being reported, please let us know what it says.

Additionally, power cycling the arm can reset the error status. Please try turning off the power completely, waiting a few seconds, and then turning it back on to see if the error persists.

Thank you.

Ottiehoblen, 2025-04-21:
Hi Yun, thanks for reaching out to us!

We did the Scan and the Wizard 2.0 only found 1 motor. I'll include the screenshots and other realtive info. 

We tried to chek the logs for any sort of error message but we didn't figure out where to do that? On a google search it says to look beneath "Packet" but that only gave us an empty CSV file. 

[image]

In addition, there are leds blinking now(at the circuit board), blue and green at a high frequency.

yun-goon, 2025-04-24:
Hello, @Ottiehoblen 

It seems there might be an issue with the cable connecting unit 1 to unit 2. Please check if the cable between unit 1 and unit 2 is properly connected.

If the issue persists, please connect the Dynamixel at unit 2 directly to the U2D2 and check the wizard results.

Additionally, to rule out a problem with the Dynamixel at unit 2, please connect the Dynamixels from unit 3 to unit 9 to the wizard as well.

closed 2025-06-20 by sunghowoo